rusefi/alphax-4chan #152: Split packages for legacy 4chan (revF and before) and current 4chan (revG and above)

ElDominio, 2023-07-29:
I updated a revF today and lost tach, which I suppose is because I had asked for #146 

Would it not be best to have two seperate packages for them? To avoid pinout problems?

My current fix:
Will downgrade to june release

rusefillc, 2023-07-29:
https://github.com/rusefi/rusefi/commit/baaec6ff328077994a03cd216287685844d2a026

so we have two options for D3 now

"D3 - Digital In 2 legacy up to rev F"
"D3 - Digital In 2 (A4 tach out up to rev F)"

ElDominio, 2023-07-29:
[image]

I can't choose tach output in outputs menu telling it that it is a revF

rusefillc, 2023-07-29:
Ohh stuff was not generating due to issue around https://github.com/rusefi/rusefi/pull/5412

rusefillc, 2023-07-29:
@ElDominio pinout and stuff is more properly generated now

https://rusefi.com/docs/pinouts/hellen/alphax-4chan/?pin=D3

ElDominio, 2023-12-05:
Pinouts between pre-RevG and RevG+ change, and my solution is to keep a July (20230705) firmware for when RevF downwards ECUs show up at my shop.

closed 2023-12-05 by rusefillc