rusefi/alphax-4chan #154: Cannot use D3 on revH

ElDominio, 2023-08-09:
[image]

Qwerty-OFF, 2023-08-09:
D3 should be H144_IN_SENS1 Gpio::F3

rusefillc, 2023-08-09:
See also https://github.com/rusefi/alphax-4chan/issues/149

```
  - pin: D3old
    # H144_IN_SENS1
    id: F3
    class: switch_inputs
    function: Digital Input 2 legacy up to rev F
    type: hall
    ts_name: D3 - Digital In 2 legacy up to rev F
    color: green

  - pin: D3
    # H144_OUT_IO13
    id: G6
    class: switch_inputs
    function: Digital Input 2. On older boards this pin is A4 tachometer out.
    type: hall
    ts_name: D3 - Digital In 2 (A4 tach out up to rev F)
    color: green
```

@Qwerty-OFF ^^^ does this look right?

rusefillc, 2023-08-09:
@Qwerty-OFF should I simply revert https://github.com/rusefi/rusefi/commit/baaec6ff328077994a03cd216287685844d2a026?

That would rase question of what the hell is https://github.com/rusefi/alphax-4chan/issues/149 about