rusefi/alphax-4chan #160: use 9924 on one of VR channels

rusefillc, 2023-08-29:


ElDominio, 2023-09-30:
bumping

9924 is a godsend on CKP, not really on CMP. Discrete is very good at decoding lower tooth cam signals, and 9924 is not so

Qwerty-OFF, 2023-12-08:
[image]
Should I move the VR input from the previously used E5/E6? Is it necessary to maintain HALL support for the third VR input? The MAX9924 allows to natively use one of the lines as a HALL input, and this functionality can be activated either by manually soldering a resistor or programmatically with a number of additional components. To use both inputs as HALL, the scheme currently used for C2/C3 can be replicated.

Qwerty-OFF, 2023-12-08:
The ability to enable pulldown on the C4/C7 lines will be removed. The pullup for use as a Hall sensor will be similar to the C2, C3, E6 lines.

closed 2023-12-09 by Qwerty-OFF