rusefi/alphax-4chan #161: 4chan digital inputs stim status

rusefillc, 2023-09-03:
trying to test the following digital inputs on rev H

```
   	engineConfiguration->triggerInputPins[0] = Gpio::H144_IN_CAM; // C7
   	engineConfiguration->triggerInputPins[1] = Gpio::H144_IN_D_4; // E6
   	engineConfiguration->camInputs[0] = Gpio::H144_IN_VSS; // C4
   	engineConfiguration->camInputs[1] = Gpio::H144_IN_D_AUX4; // E7
   	engineConfiguration->camInputs[2] = Gpio::H144_IN_SENS2; // E3
   	engineConfiguration->camInputs[3] = Gpio::H144_IN_SENS3; // E4
```

using stim board

four lines work, the two failing lines are ``Gpio::H144_IN_D_4; // E6`` and ``Gpio::H144_IN_D_AUX4; // E7``

* anything special about those two lines?
* what other digital input pins can I test using STIM board?

rusefillc, 2023-09-03:
```
* ZERO Secondary event counter!
* ZERO VVT2 event counter!
```

Qwerty-OFF, 2023-09-03:
Check that the pullup is enabled for E6. C2 and C3 are similar to E6, try them too.
Schematic for E7 looks good, check that there is a pull-up to 5V (R85)
Also you can test D3, F3.