rusefi/alphax-4chan #162: document ETB capacitors

rusefillc, 2023-09-04:
https://github.com/rusefi/alphax-4chan/issues/110 has added electrolytic while not removing ceramic capacitors

we have never removed electrolytic from BOM even while https://github.com/rusefi/alphax-4chan/issues/110 was implying only footprint for custom board manual modification

At the moment all recent 4chan were fabricated with both sets? Is that a bug or a feature? Let's add text comment on the ETB page.

rusefillc, 2023-09-04:
let's move optional (?) electrolytic capacitor from the reusable reusable page to 4chan main page and mention that it technically is not needed

Qwerty-OFF, 2023-09-04:
Done, electrolytic capacitor is now DNP
[image]

closed 2023-09-04 by Qwerty-OFF