rusefi/alphax-4chan #166: Make Battery Voltage read from A2 (main relay)  instead of F7

ElDominio, 2023-10-23:
[image]
Pictured: Key input has 11.3v when battery has 12.1v

Using key input as battery voltage is incorrect since we base our Dwell and Injection dead time corrections off battery voltage, we should read from the same source. On most cars, key-on is usually a bit smaller voltage than main relay, but sometimes even 1v off.

rusefillc, 2023-10-25:
@ElDominio this is harder than just a hardware change. At the moment we engage main relay when we sense voltage on ignition key.

if we only make the hardware change we would have to adjust 4chan settings to ALWAYS engage main relay which I assume is not desired.

For 4chan to read voltage from main relay we would need to ADD analog input not just MOVE.

@ElDominio looks like we would have to reuse one of aux analog inputs for main relay voltage. I suggest https://rusefi.com/docs/pinouts/hellen/alphax-4chan/?pin=X3

@ElDominio please confirm that we repurpose X3 analog input for main relay input

ElDominio, 2023-10-25:
Can we not sense ignition ON voltage with a digital pin, then use analog to read voltage from A2?

ElDominio, 2023-11-04:
I made a circuit to do this in one of my boards, if you want i can post it here (external main relay control)

ElDominio, 2023-11-09:
[image]

just in case

rusefillc, 2023-12-05:
[image]

rusefillc, 2023-12-07:
@ElDominio I do not have brain capacity to deal with this until January.

Do you want "rev-i" without this ticket in Dec or with it in January?

ElDominio, 2023-12-07:
With it in January

rusefillc, 2023-12-10:
```
#if EFI_MAIN_RELAY_CONTROL
	hasIgnitionVoltage = Sensor::getOrZero(SensorType::BatteryVoltage) > 5;
```

rusefillc, 2023-12-10:
Let me remove R53 to test https://github.com/rusefi/rusefi/commit/3b88f847e0370de6afc359f88c7c315cf695689c

open question what pin to use for ignitionKeyDigitalPin

rusefillc, 2023-12-10:
[image]

Qwerty-OFF, 2023-12-10:
Схема не подойдет для вариантов где всегда (на заглушенной машине) есть напряжение на нашем vign входе, будет разряжать аккумулятор. Также измерение напряжения главного реле через резисторный делитель тоже будет расходовать заряд батареи. Для автомобилей у которых питание always hot и запуск по can wakeup надо придумать что-то лучше.

rusefillc, 2023-12-12:
устал я от этой темы, я уже согласен на второй аналоговых вход

а можем ли мы сделать AV из AT1 например удалением резистора?

[image]

Qwerty-OFF, 2023-12-12:
Да, можно сделать av через replace

rusefillc, 2023-12-12:
@Qwerty-OFF давай заведём main relay +12v на AT1 с делителем

closed 2023-12-16 by Qwerty-OFF