rusefi/alphax-4chan #173: Can High side Outputs drive VVT solenoids

ElDominio, 2023-12-20:
Ask question states, can high side output drive a VVT solenoid that has one side wired to ground?

rusefillc, 2023-12-20:
@ElDominio please provide context what pins are you asking about? ignition control I am sure is too weak

ElDominio, 2023-12-20:
Internal High1 and High2, the ones on the molex microfit

Qwerty-OFF, 2023-12-20:
Yes, we need an experiment.  Perhaps long-term operation with a current of 3 Amps will cause significant heating.  These outputs are unprotected.  Flyback diodes are present, you can control the inductive load using PWM.

ElDominio, 2023-12-20:
I don't think VVT solenoids ask for more than half an amp, so we should be fine?

rusefillc, 2023-12-20:
> we need an experiment

@ElDominio would you be able to make an experiment happen?

ElDominio, 2023-12-20:
What is the experiment

rusefillc, 2023-12-20:
@ElDominio please wire up a VVT solenoid on bench or real car and see what happens

ElDominio, 2024-02-23:
This will be tested eventually

ElDominio, 2024-05-16:
It may not be because #176 happened