rusefi/alphax-4chan #175: Request: Can 4chan firmware build have things that will never be used REMOVED?

ElDominio, 2024-02-21:
Recent updates to main RusEFI branch have removed LUA space which has caused issues when updating cars running with Lua

From what I can see, none of the transmission control stuff is applicable to 4chan
[image]

Could this be removed to make me happy in the Lua?

rusefillc, 2024-02-21:
@ElDominio this could save you maybe 100 bytes. The level of effort required to save you relatively few bytes would not make sense for me unfortunately.

ElDominio, 2024-02-21:
Right now there's not much info on what makes scripts larger/smaller besides the "x bytes used after rundown"; and after succesful build there's no indication of total vs used space. My old Lancer scripts do not build on latest firmware. I updated one from 20230706 to 20240219 and reloaded firmware from tunerstudio and CAN was broken. If nothing can be done, then the issue can be closed.

rusefillc, 2024-02-21:
@ElDominio with firmware progress we get more and more system functions which reduce space available to users

I've bumped 4chan lua heap a bit please give it a try

closed 2024-02-21 by rusefillc