rusefi/alphax-4chan #182: M54 wired as PNP has odd issues in cylinders 5 and 6

ElDominio, 2024-07-10:
Tune and log: https://rusefi.com/online/view.php?log=1411

The car fires injector and ignition 5 and 6, but unplugging them with engine running results in no change; the engine backfires sometimes, which is strange?
Coil 1 and 3 are swapped physically, so that's why the pins are swapped in firmware. Testing coils and injectors 1-6 is on the correct cylinders according to numbers on the Bench Test area.

It sounds as if 5 and 6 are firing in the wrong place, but they fire correctly in test mode? I am incredibly confused

rusefillc, 2024-07-12:
please remove fly back diodes?

please convert to batch fuel if above does not help?

[image]

[image]

rusefillc, 2024-07-12:
@ElDominio maybe those

[image]

rusefillc, 2024-07-12:
[image]

ElDominio, 2024-07-12:
Flyback diodes were the problem, removed and we have full power

rusefillc, 2024-07-12:
@ElDominio what's the next step for this ticket?

ElDominio, 2024-07-12:
Closing it? Remove it from default BOM?

rusefillc, 2024-07-12:
@ElDominio yes we can remove a couple of default BOM if you need

ElDominio, 2024-07-24:
removing diode was fix so i think it can be closed