rusefi/alphax-4chan #185: VVT solenoid bank 1 intake is missing

ElDominio, 2024-07-24:
[image]

how

rusefillc, 2024-07-24:
@ElDominio what pin are we talking about? https://rusefi.com/docs/pinouts/hellen/alphax-4chan/?pin=F2 or https://rusefi.com/docs/pinouts/hellen/alphax-4chan/?pin=F4 ?

please clarify what do I do wrong see below

[image]

ElDominio, 2024-07-24:
My screenshot is missing "VVT solenoid bank 1 intake"
this is on 20240723

pin F2, but the issue is that I can't assign it
I was able to assign in
[image]

but it should be showing on vvt